pololu/vl53l1x-arduino #1: Sensor printing 0's

wightga, 2018-06-14:
When trying to run the continuous sketch or the other continuous sketch, the serial monitor just shows these being printed out.
⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮

The sensor works because it has printed out values on a different code from a different library and there must be something wrong with this one but I can't seem to find the issue.

ruimarinho, 2018-06-14:
I think that’s a question of adjusting the baud rate of the serial monitor. 115k.

wightga, 2018-06-14:
Yep, I literally just changed that right before I saw your answer and it worked. What exactly is the baud rate? and what does it do?

ruimarinho, 2018-06-14:
It’s the number of bits per seconds transmitted by the communication channel. If it is incorrect, the monitor won’t be able to extract the information being sent.

closed 2018-06-14 by ryantm