pololu/vl53l1x-arduino #3: Single Shot Measurements - Feature Request

JGSuw, 2018-07-24:
Hello, 

First, thank you for producing this API. I quite like it because it is straightforward and implements many of the core configurations of the sensor (namely, range mode and timing budget).

I noticed in the comments there is a reference to a single-shot measurement function, but it appears to not yet been implemented. I believe that the repository would benefit from a function which allowed the user to initiate a single measurement; that way, once could read the measurement after the timing budget has passed and have more certainty about when the measurement occurred. 

Do you have plans to implement this feature?

kevin-pololu, 2018-08-06:
Hi,

At the time I was working on this library, the only example programs I could find for ST's API came from [X-CUBE-53L1A1](https://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32cube-expansion-software/stm32-ode-sense-sw/x-cube-53l1a1.html) and used continuous measurements, and I don't remember finding enough documentation about single-shot measurements to be confident about implementing them, but I'll try to look again sometime to see if anything has changed.

The reason that the comments mention single-shot measurements is that I accidentally left that in there after using our VL53L0X library as a starting point.

Kevin

OpenUAS, 2019-11-15:
Would be a welcomed addition, since a VL53L0X works perfectly connected to a ESP8266 in single measurement on own interval. in contrast the VL53L1X wifi init is problematic due to resource timeing budget with the continuous mode on the VL53L1X lib sadly. @JGSuw the code from the VL53L0X lib could be added to the VL53L1X lib with no all to much effort I think. @JGSuw if you are lucky will try to add the code if I do no use the Sparkfun VL53L1X lib in the end... sincd I do not think @kevin-pololu will add it looking at the date of this isssue...

ruimarinho, 2019-11-15:
I'm facing a strange issue where after about two weeks of running on an ESP8266, I start getting signal failures. The sensor doesn't physically move a bit, so I am thinking this has to be a software issue. Any idea of this could be related due to the use of continuous mode?

[image]

OpenUAS, 2019-11-16:
Best to make a new issue please.

kevin-pololu, 2020-09-22:
I know this feature request has been open for a while now, but we recently encountered a situation where being able to do single-shot measurements would be useful, and I decided to look into this some more. I still don't think I've found any example code or documentation from ST that shows how to do single-shot measurements, but it turned out to be pretty straightforward to copy the same approach used in ST's VL53L0X API and our VL53L0X library (basically just writing to a different start bit to trigger a single measurement instead of continuous measurements), and it seems to work fine as far as I can tell.

The latest library release (1.1.0) adds a function for doing a single-shot reading; if anyone is still looking for this functionality, please try it out and let me know if you find any problems.

closed 2020-09-22 by kevin-pololu