pololu/vl53l1x-arduino #7: Failed to detect sensor

Gbouna, 2019-06-06:
Thank you for this example it is really helpful. However, when i uploaded the code, opened my serial monitor, it shows "Failed to detect and initialize sensor!". What could be the possible problem why the sensor failed to be detected and initialized. 
Thank you

kevin-pololu, 2019-06-12:
That typically means the Arduino is unable to communicate with the sensor (for example, because of bad connections). If you are still having trouble with this, please post to [the Pololu forum](https://forum.pololu.com/), since that is a more appropriate place for troubleshooting. It would be helpful if you include pictures of your setup so we can check your wiring and soldering.

closed 2019-06-12 by kevin-pololu

Gbouna, 2019-06-13:
okay thank you, i will do just that