pololu/vl53l1x-arduino #8: Compiler problem

jps2000, 2019-06-24:
got a c ompiler error:   call of overloaded 'write(int)' is ambiguous
Cure: 
Change line 558 in .cpp file from
 Wire.write((RESULT__RANGE_STATUS >> 8) & 0xFF)); // reg high byte
to
  Wire.write((uint8_t)((RESULT__RANGE_STATUS >> 8) & 0xFF)); // reg high byte
then it worked

may depend on wire library???

kevin-pololu, 2019-07-23:
What kind of Arduino are you using (compiling for)?

jps2000, 2019-07-24:
IDE Arduino 1.8.5 
Board is Adafruit Bluefruit nrf 52832 Feather
Thanks

kevin-pololu, 2019-07-30:
When I tried compiling for the Feather nRF52832, I wasn't able to reproduce your problem with Arduino 1.8.9 and version 0.11.1 of the "Adafruit nRF52" boards package (both of the example sketches compiled without errors). Could you see if updating to 1.8.9 resolves the problem for you with the unmodified library? If not, could you post the full console output (containing the error message) that you get when you try to compile?

jps2000, 2019-07-31:
Hi Kevin,
Here the full console output
Arduino: 1.8.5 (Windows 7), Board: "Adafruit Bluefruit nRF52832 Feather, 0.2.6 (s132 6.1.1), Level 0 (Release)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Xyz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Xyz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Xyz\Dropbox\arduino_sketchbook\libraries -fqbn=adafruit:nrf52:feather52832:softdevice=s132v6,debug=l0 -ide-version=10805 -build-path C:\Users\XYZ~1\AppData\Local\Temp\arduino_build_473653 -warnings=default -build-cache C:\Users\XYZ~1\AppData\Local\Temp\arduino_cache_314658 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.nrfjprog.path=C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\nrfjprog\9.4.0 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4 -verbose C:\Users\Xyz\Dropbox\featherNRF52\DOFTOF_V1\DOFTOF_V1.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Xyz\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Xyz\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Xyz\Dropbox\arduino_sketchbook\libraries -fqbn=adafruit:nrf52:feather52832:softdevice=s132v6,debug=l0 -ide-version=10805 -build-path C:\Users\XYZ~1\AppData\Local\Temp\arduino_build_473653 -warnings=default -build-cache C:\Users\XYZ~1\AppData\Local\Temp\arduino_cache_314658 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.nrfjprog.path=C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\nrfjprog\9.4.0 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4 -verbose C:\Users\Xyz\Dropbox\featherNRF52\DOFTOF_V1\DOFTOF_V1.ino
Using board 'feather52832' from platform in folder: C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3
Using core 'nRF5' from platform in folder: C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3
Detecting libraries used...
"C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=64000000 -DARDUINO=10805 -DARDUINO_NRF52_FEATHER -DARDUINO_ARCH_NRF52 -DARDUINO_BSP_VERSION="0.9.3"  -DNRF52832_XXAA -DNRF52 -DARDUINO_FEATHER52 -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DS132 -DCFG_DEBUG=0 -Os "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/cmsis/include" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/nrfx" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/nrfx/hal" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/nrfx/mdk" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/nrfx/soc" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/nrfx/drivers/include" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/nordic/softdevice/s132_nrf52_6.1.1_API/include" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/freertos/Source/include" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/freertos/config" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/freertos/portable/GCC/nrf52" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/freertos/portable/CMSIS/nrf52" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/sysview/SEGGER" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/sysview/Config" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/usb" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5/usb/tinyusb/src" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\cores\nRF5" "-IC:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\variants\feather_nrf52832" "C:\Users\XYZ~1\AppData\Local\Temp\arduino_build_473653\sketch\DOFTOF_V1.ino.cpp" -o "nul"
"C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ [... cut ...]
Using library SPI at version 1.0 in folder: C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\libraries\SPI 
Using library arduino_404169 at version 1.1.8 in folder: C:\Users\Xyz\Dropbox\arduino_sketchbook\libraries\arduino_404169 
Using library FileSystem at version 0.9.3 in folder: C:\Users\Xyz\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.9.3\libraries\FileSystem 
**exit status 1
call of overloaded 'write(int)' is ambiguous**

Have not  tried updated software.

kevin-pololu, 2019-07-31:
I found that downgrading my nRF52 boards package to 0.9.3 (which you are using) caused the compilation error to occur, and upgrading back to the latest 0.11.1 made it compile fine again. Could you try upgrading the boards package and see if that fixes it for you?

jps2000, 2019-08-01:
Yes that worked- but had to modify others unfortunately.
Also tried 
Wire.write((uint8_t)((RESULT__RANGE_STATUS >> 8) & 0xFF)); // reg high byte
with 0.11.1 and  it worked also . The latter is more generic though.

closed 2019-08-01 by jps2000