pololu/vl53l1x-arduino #10: Calibration functions

agrath, 2019-09-17:
I am looking at using a VL531X sensor (well, many) for a project.

Our application will require cover glass to be placed over the sensor, which means we will likely need to re-calibrate the sensor as per the manual [UM2356](https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/98/0d/38/38/5d/84/49/1f/DM00474730/files/DM00474730.pdf/jcr:content/translations/en.DM00474730.pdf), section 3.

According to the manual, the following functions are required:
- VL53L1_PerformRefSpadManagement
- VL53L1_PerformOffsetSimpleCalibration
- VL53L1_PerformSingleTargetXTalkCalibration
- VL53L1_SetXTalkCompensationEnable
- VL53L1_GetCalibrationData
- VL53L1_SetCalibrationData

Plus the data structure for calibration data, VL53L1_CalibrationData_t and VL53L1_customer_nvm_managed_t (containing the 8 RefSpad calibration parameters)

You need to call the first 3 functions in order, with a gray target (17 % reflectance at 940 nm) placed at measured 140mm for the VL53L1_PerformOffsetSimpleCalibration call, then call VL53L1_GetCalibrationData to obtain the values. 
These values can then be re-loaded at boot using VL53L1_SetCalibrationData after VL53L1_DataInit and VL53L1_StaticInit (which appear to be unrolled into init)

Cross talk compensation must also be enabled using VL53L1_SetXTalkCompensationEnable if you're using cover glass.

It appears from reviewing the source that the init function of this library presently applies static config calibration data around line 100, however I'm not sure if this additional calibration data from VL53L1_SetCalibrationData is the same.

Would it be possible to have support for these functions added to the library? As far as I can tell, this would make it feature complete.

agrath, 2019-09-17:
Note: https://github.com/pololu/vl53l1x-st-api-arduino/blob/master/vl53l1x-st-api/vl53l1x-st-api.ino provides a way to directly call the API functions rather than using this wrapper, however that looks like a lot of pain that could be avoided

hennieb, 2020-06-19:
> I am looking at using a VL531X sensor (well, many) for a project.
>
> Our application will require cover glass to be placed over the sensor, which means we will likely need to re-calibrate the sensor [...]

Hello, did you get this resolved? I have a similar scenario and need to calibrate the device with a cover glass in place.  Please let me know how you got on with the calibration process.

Lyr3x, 2021-09-23:
That would be really good to have in this library as it has most of the advanced features implemented then, with a usable API. Including the complete API comes with much bloat. That’s why I implemented this library  in my project. 
Users keep asking for cover glass calibration so I am kind of stuck here as well. 
@kevin-pololu do you intend to implement the missing functions?
I’ll check if I could provide a PR