pololu/vl53l1x-arduino #12: Measure 2 distances, same sensor

elpedriyo, 2019-10-28:
Hello,

I have checked this video: https://www.youtube.com/watch?v=c91Ve-g0J2U for the sensor vl53l1x that was created by STMicroelectronics and it seems they use some kind of the same sensor to take 2 different readings I guess, do you have any clue on how could I achieve this with your library, or how are they doing it?

Kind regards

Shamshala, 2019-10-28:
I think they use SPADs to determine the direction.

kevin-pololu, 2019-10-30:
It looks like you also asked about this [on the Pololu forum](https://forum.pololu.com/t/measure-2-distances-same-sensor-vl53l1x/18212), where we responded. For the benefit of any other readers, here is a copy of that response:

> We haven't tried people counting with the VL53L1X, but it looks like TI has a document (UM2600) which is titled "Counting people with the VL53L1X long-distance ranging Time-of-Flight sensor" that would be a good place to start.  That can be found on [ST's product page](https://www.st.com/en/imaging-and-photonics-solutions/vl53l1x.html#resource) for that sensor, along with other resources.  Note that our library does not support some functionality that is likely needed for this application (such as setting a user-defined ROI, or region of interest), so you would probably have to use the API provided by ST.
>
> For additional information, you could try broadening your search to include places like the Arduino forums, which seems to have [discussions like this one](https://forum.arduino.cc/index.php?topic=621855.0).  Finally, you might also consider posting a question on the [ST community site](https://community.st.com/s/topic/0TO0X000000BSqxWAG/proximity-sensors).

(Note that we have also written an [implementation of ST's VL53L1X API for the Arduino platform](https://github.com/pololu/vl53l1x-st-api-arduino/) that could be useful for this.)

closed 2019-10-30 by kevin-pololu