pololu/vl53l1x-arduino #15: About vl53l3cx

Ray0124, 2020-06-17:
Hi pololu,
Would you transplant vl53l3cx-api into  vl53l3cx-api-arduino in the future?
Or could you tell me some hints to do this,I want to use vl53l3cx-api  on arduino, but I don't know how to do this. Its api some part is similar to vl53l1x.

kevin-pololu, 2020-08-07:
Hi, Ray0124.

We are looking into producing a carrier board for the VL53L3CX and investigating what kind of software support we might provide for it. Based on what we have learned so far, it seems like it will be impractical to run the VL53L3CX API on an 8-bit Arduino because the multi-target detection processing requires a large amount of RAM (on the order of 10 KB). However, it might still be possible to port the API to the Arduino environment for use with 32-bit boards with more resources.

I'm closing this issue because it's not a bug or feature request related to our VL53L1X library (for future reference, our [forum](https://forum.pololu.com/) would be a more appropriate place for this kind of discussion), but we are interested in hearing why you want to use the VL53L3CX instead of another ST ToF sensor (is the multi-target detection an important feature for you?) and what kind of Arduino board you want to run its API on. You can briefly reply here, or if you expect to start an extended discussion, please create a topic on our forum.

Kevin

closed 2020-08-07 by kevin-pololu

Ray0124, 2020-08-11:
Thanks for replying!! Oh, I am  interested in the feature " the multi-target detection " of the  VL53L3CX , because I  want to combine it with some of my projects on arduino UNO . I just want to do some sensors to let the machines  know 
obstacles and the distances of obstacles  around  them and they can simply do something to response according to distances . Thanks for reminding me of the problem of the ram ,and I want to change it on arduino DUE  or other controllers.  OK , I would take time to create a topic on your forum after I trying something with VL53L3CX  API and my projects on  arduino DUE or Max32630. Thank you!