pololu/vl53l1x-arduino #19: error: using typedef-name 'TwoWire' after 'class'

andbe-gh, 2020-11-23:
Hi,

with updating to version 1.2, my code stopped working with the following error message in the IDE:

```In file included from C:\Users\xxx\Documents\Arduino\libraries\xxx/xxx.h:29:0,
                 from C:\Users\xxx\Documents\Arduino\xxx\xxx.ino:59:
C:\Users\xxx\Documents\Arduino\libraries\VL53L1X/VL53L1X.h:5:7: error: using typedef-name 'TwoWire' after 'class'
 class TwoWire; // from Wire.h
       ^~~~~~~
In file included from C:\Users\xxx\Documents\Arduino\libraries\Arduino_LSM9DS1\src/LSM9DS1.h:21:0,
                 from C:\Users\xxx\Documents\Arduino\libraries\Arduino_LSM9DS1\src/Arduino_LSM9DS1.h:23,
C:\Users\xxx\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\libraries\Wire/Wire.h:86:26: note: 'TwoWire' has a previous declaration here
 typedef arduino::MbedI2C TwoWire;
                          ^~~~~~~
```
When I downgrade to 1.1, the code is working again. Any ideas?

closed 2020-11-23 by kevin-pololu

kevin-pololu, 2020-11-23:
That's related to the changes we made to support different I2C buses, but we didn't consider the possibility that `TwoWire` might be defined differently on some platforms (like mbed, which it looks like you're using). I just made a 1.2.1 update that should fix this; please try it and let us know if you still run into problems.

andbe-gh, 2020-11-24:
Oh sorry, I've forgotten to mention that. Yes, I'm using an arduino nano ble sense with mbed.
Good news, compiling works again after updating to 1.2.1. Thanks.

commit dcee24af (2020-11-23): include Wire.h from VL53L1X.h

instead of just declaring TwoWire class, which causes problems on
platforms where TwoWire isn't a class - fixes #19