pololu/vl53l1x-arduino #24: VL53L1X sensor breakdown after a few minutes

hsynklceng, 2023-08-10:
Hello, I'm trying to use 12 pololu vl53l1x(ToF) sensors with the stmf407vgt6 custom card. In addition, this board has a 5A 2 channel motor driver. I think the i2c data is getting noise during the motor start cycle. But according to my observation on debug mode, ToF sensor is break down. Randomly some of the sensors break down. How can i fix this problem?

kevin-pololu, 2023-08-11:
Hi, can you explain what you mean by "break down"? What debug information do you see that makes you think that is happening?

hsynklceng, 2023-08-11:
I mean it is closing it self. Sensors stop send to data when it starts motor. Actually not noise ToF stops to send data.

kevin-pololu, 2023-08-12:
Are you using our Arduino library from this repository? The variables you are watching look more like the ones from ST's VL53L1X API.

presslab-us, 2024-05-02:
I'm also having the problem where the sensor stops reporting good data after some minutes time.  It seems to vary from 30 seconds to 5 minutes.  I'm not sure what causes the variation.

I've since switched to using the library here and so far it's gone a whole day without stopping, I expect that the problem is fixed.
https://github.com/stm32duino/VL53L1X

kevin-pololu, 2024-05-07:
@presslab-us, thanks for the information. If you do any further troubleshooting and find out more that might help us narrow down the source of the problem, please feel free to open a new issue. I'm going to close this one because we never confirmed whether the original submitter was using our library at all.

closed 2024-05-07 by kevin-pololu